glowbuzzer/gbr #33: show frames on 3d robot view

jugglingcats, 2022-08-25:


jugglingcats, 2022-08-25:
Think this needs some more discussion. Random thoughts...

- We can't have a simple "stack" of frames visually which are clickable, like on chilipeppr, because we have a tree hierarchy
- What happens when I select a frame? We can show a triad at the frame's origin, but does it affect anything
- Possible confusion between G5x drop down on the gcode tile. Again, what does selecting a frame in toolpath tile actually do?
- Maybe it's more a visual frame "inspector"... tick a box to enable inspector: show feint frame triads and frame hierarchy in the corner, then mouse over to highlight each one

deeegee, 2022-09-07:
Yes it is absolutely the inspector rather than the clickable frames

closed 2022-11-24 by jugglingcats